KlausMu/esp32-fan-controller #1: control multiple fan support

thomasesr, 2021-03-05:
Hello. It would be nice to have support for multiple fans since the ESP32 Hardware is capable enough.

KlausMu, 2021-03-05:
How would you like to control multiple fans?
- "temperature controller mode" or "pwm mode"?
- all with the same target temperature, or different target temperatures?
- using only one temperature sensor for all fans, or each fan with its own temperature sensor?

Note that it should be possible to simply connect the pwm signal to more than one fan to let them run all at the same speed.

There are so many combinations I can think of, I don't know if this can easily be achieved in a generic way. Maybe it would be better simply to use my project as a starting point and to adapt it to your needs?

lrascao, 2021-03-05:
that's actually what i did, i'm controlling 8 PWM fans using mqtt in my project. Your project helped a lot! Thanks!

thomasesr, 2021-03-05:
> How would you like to control multiple fans?
>
>     * "temperature controller mode" or "pwm mode"?
>
>     * all with the same target temperature, or different target temperatures?
>
>     * using only one [...]

My thinking was having multiple pwm signals and multiple tacho inputs and maybe just a simple 4 point temperature curve you can configure for each (hardcoded even). Similarly of what you can find in BIOS fan control of motherboards (not the really expensive ones).

KlausMu, 2021-03-05:
Sure, this could be done

- see "temperatureControll.cpp" for creating a 4 point temperature curve

- almost all the other files have to be adjusted so that not only one fan/temperature sensor is recognized, but many. But this should be quite easy to do. At least for the sensors and fan control itself. Bringing all these information to the display is maybe the hardest part, because there is not so many space left :-)

If you have any questions, feel free to ask.

closed 2022-01-29 by KlausMu