KlausMu/esp32-fan-controller #2: DS18B20 Temperature probe

thomasesr, 2021-03-05:
It would be nice to also have the DS18B20 (1-wire) temperature sensor since you can have multiple sensors in the same IO pin you would be able to have a temperature sensor for each Fan and an ambient temperature sensor.

KlausMu, 2021-03-05:
There are so many temperature sensors out there, it is impossible to support all of them.
It should be very easy to adapt to code to support other sensors than the BME280 used by me.
Just have a look at "sensorBME280.h" and "sensorBME280.cpp".

lrascao, 2021-03-05:
biggest advantage of DS18B20 is that it uses the 1-wire protocol, with 1 GPIO you can read from several of these

KlausMu, 2021-03-05:
Ok, I see. The BME280 sensors uses I2C, so with only 2 GPIOs you can also have up to 127 sensors.

thomasesr, 2021-03-05:
DS18B20 its way cheaper, specially for a simple fan control and you only need to monitor temperature and not pressure and humidity. Also it has a wider range of temperature to measure. Which is nice.

tricoos, 2021-09-03:
The main advantage for me is that the DS18B20 is available with a longer cable and waterproof so you can position it way easier than the non-waterproof BME board.

closed 2022-01-29 by KlausMu