KlausMu/esp32-fan-controller #8: incorrect data

mkeyno, 2022-01-26:
Hi 
thanks for sharing the code, I've tried to use your code and pair of two PNP proximity sensors to measure 2 motor speeds, but I find if I decrease the `tachoUpdateCycle` time , it shows the incorrect speeds, 
I was thinking it may be a better idea to use a time span between the pulses instead of counting the pulses

KlausMu, 2022-01-26:
Oh, it seems there is a bug in the code. `tachoUpdateCycle = 1000`  means you are counting the pulses for one second. Afterwards, when calculating rpm (revolutions per minute) out of the counted pulses, you have to take into account how long you did the measurement.
So probably instead of
`last_rpm = counter_rpm * (60 / numberOfInterrupsInOneSingleRotation);`
it should be
`last_rpm = counter_rpm * (60 / numberOfInterrupsInOneSingleRotation) * (tachoUpdateCycle / 1000);`
Could you please try this with different tachoUpdateCycles? Should always result to the same rpm.

In case it is not clear: the pulses are counted continuously. `tachoUpdateCycle` only means how often you calculate the rpm out of the counted pulses and reset the pulse counter `counter_rpm`. I think this is the best way to do it. Measuring the time span leads to other problems, e.g. if the time span always differs a litte bit. The interrupt routine has to be as simple as possible in order to take as less time as possible, and there is nothing simpler than `counter_rpm++;`

KlausMu, 2022-01-30:
I did this change in the code and commited it.

closed 2022-01-30 by KlausMu