KlausMu/esp32-fan-controller #9: Question re pullup resistors

patrickcollins12, 2022-01-29:
Question for you. Is there a reason why are you using pull-up resistors when the esp32 has built-in pull-up resistors?

KlausMu, 2022-01-29:
Yes, I explained it here: [https://github.com/KlausMu/esp32-fan-controller/wiki/02-Obstacle-ESP32-electronics](https://github.com/KlausMu/esp32-fan-controller/wiki/02-Obstacle-ESP32-electronics)

You can try without pullups, if it works for you, then you are lucky.

For the tacho you have to change
`  pinMode(tachoPin, INPUT);`
to
`  pinMode(tachoPin, INPUT_PULLUP);`

For I2C the internal pullups are active anyway (done by the library used).

patrickcollins12, 2022-01-29:
Ha, well there you go. This might explain why I couldn't get an RPM readout from a hall sensor on my gokart. Will have to go back and look into it.

I've just built a very similar project to yours using esphome. I'll send you a link if you're interested.

KlausMu, 2022-01-29:
> I've just built a very similar project to yours using esphome. I'll send you a link if you're interested.
Sure.

closed 2022-02-04 by KlausMu

patrickcollins12, 2022-02-06:
I Klaus, I finally got around to take some photos and writing the README. I was inspired by your write-up. I'd love your feedback.

https://github.com/patrickcollins12/esphome-fan-controller

KlausMu, 2022-02-06:
Hi Patrick, thanks for the link, looks great! Thanks for publishing it. Here are some points you might like to consider:

- Duty cycle vs. voltage: You are talking about "the % of voltage being sent out via PWM". This is misleading, because this kind of fan is not controlled via voltage, but via pwm. The voltage to power the fan is always 12 V. The pwm pin from the ESP32 which controls the speed is always using 3.3 V or 0V. The speed of the fan is controlled via the duty cycle of the pwm signal. You never have any other voltage than 0, 3.3 and 12. See the last picture here https://github.com/KlausMu/esp32-fan-controller/wiki/02-Obstacle-ESP32-electronics

- Fan control signal: Your control signal is connected to pin 3 of the fan, but it should be pin 4. Pin 3 is the tacho signal.

- LM2596: do you really need the connection between IN- and OUT-? A quick test showed me that they are already connected.
- LM2596: wiring diagram has a minor flaw. GND is connected to IN+

- DHT11: I know that there exist different versions with different pinouts of the DHT11. I indeed have a 4-pin, which uses a different pin for GND than in your wiring diagram (see https://github.com/KlausMu/esp32-co2monitor)