KlausMu/esp32-fan-controller #11: Compilation Error

anaegeli, 2022-08-25:
Hi, 

I got a compilation error on my board.
I didn't know if the problem is my board, or if i'm missing something.

```
src/wifiCommunication.cpp:126:70: error: no matching function for call to 'WiFiClass::onEvent(void (&)(arduino_event_id_t, arduino_event_info_t), system_event_id_t)'
   WiFi.onEvent(WiFiStationDisconnected, SYSTEM_EVENT_STA_DISCONNECTED);
                                                                      ^

src/wifiCommunication.cpp:155:64: error: no matching function for call to 'WiFiClass::onEvent(void (&)(arduino_event_id_t, arduino_event_info_t), system_event_id_t)'
   WiFi.onEvent(WiFiStationConnected, SYSTEM_EVENT_STA_CONNECTED);
                                                                ^
src/wifiCommunication.cpp:156:50: error: no matching function for call to 'WiFiClass::onEvent(void (&)(arduino_event_id_t, arduino_event_info_t), system_event_id_t)'
   WiFi.onEvent(WiFiGotIP, SYSTEM_EVENT_STA_GOT_IP);
                                                  ^
src/wifiCommunication.cpp:157:70: error: no matching function for call to 'WiFiClass::onEvent(void (&)(arduino_event_id_t, arduino_event_info_t), system_event_id_t)'
   WiFi.onEvent(WiFiStationDisconnected, SYSTEM_EVENT_STA_DISCONNECTED);
                                                                      ^

```

Any idea how to fix this?
Best regads,
Adrian

KlausMu, 2022-08-27:
Seems the new version of the Espressif32 platform broke the API. I have to dig into this deeper.
As a workaround, please change in file "platform.ini"
this line
`platform = espressif32`
into this line
`platform = espressif32@^3.5.0`

anaegeli, 2022-08-27:
Here my actual platformio.ini
`
[env:esp32-c3-devkitm-1]
platform = espressif32@^3.5.0
board = esp32-c3-devkitm-1
framework = arduino
monitor_speed = 115200
upload_port = COM6
lib_deps = 
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit BME280 Library@^2.1.2
	adafruit/Adafruit BusIO@^1.7.0
	adafruit/Adafruit ILI9341@^1.5.6
	paulstoffregen/XPT2046_Touchscreen@0.0.0-alpha+sha.26b691b2c8
	gerlech/TouchEvent@^1.3
	adafruit/Adafruit ST7735 and ST7789 Library@^1.6.0
	jandrassy/TelnetStream@^1.2.1
`

And here the error that i get now:
`
Processing esp32-c3-devkitm-1 (platform: espressif32@^3.5.0; board: esp32-c3-devkitm-1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via '-v, --verbose' option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
Error: This board doesn't support arduino framework!
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv-esp @ 1.80400.0 (8.4.0)
================================================================================================= [FAILED] Took 0.91 seconds =================================================================================================
`

KlausMu, 2022-08-28:
The ESP32-C3 board wasn't supported in version 3.5.0 of the esp32 arduino core.

Then you have to do more changes.

In file "wifiCommunication.cpp", replace all occurences of
`SYSTEM_EVENT_*`
with
`ARDUINO_EVENT_WIFI_*`
You have to do this four times.

In file "platform.ini":
`platform = espressif32`
`board = esp32-c3-devkitm-1`

In file "config.h":
`const uint32_t I2C_FREQ             = 100000; // 400000`

Note: since the ESP32-C3 is pin compatible with ESP8266, but not with with ESP32, you have to do more changes in "config.h". GPIO 22, 23, 32 and 33 are not available. See [here](https://socialcompare.com/en/comparison/esp8266-vs-esp32-vs-esp32-s2)

closed 2022-09-02 by anaegeli