KlausMu/esp32-fan-controller #18: updates, project.

s4saken, 2023-11-25:
Hi Klaus, 

I see your very busy making this great. Thanks man for all the effort. 
I got a new touch display and its working great again. 

I managed to setup new pwm channels to. 
Because i have 2 fans for my case. I wanted to create a positive pressure for in/out. So i got this working on a second temp control, but with an offset. Kinda great.  

The second channel i use for a build in filter, for when i dont need any temp control. 
Also bought a 2 channel relais module. Gonna create 2 switches for light and case power. 
I am gonna add the mqtt discovery messages for these also. 

All and all, very nice project, nice base and just a very good smart device. 

Testing the new updates right now.

Kind Regards,

closed 2023-11-25 by s4saken

KlausMu, 2023-11-25:
Thanks a lot for your feedback. I really appreciate it.
I'm finished now with all the planned changes. Nothing more on the to do list.
So you can test witout fear that something new is in the pipeline.

s4saken, 2023-11-25:
Well i see you added something new in the pwm control also, regarding setting the off state on pwm controlled fans. I guess this is a fix for the problem i was looking for, since my self added pwm channels, did not comply with off state. Gonna test is right now :D. Thanks again.

KlausMu, 2023-11-25:
Well, mainly I added an explicit "off" state, inspired by how HA is doing it.
If the device is "off", pwm is set to 0, and also kept at 0, no matter what the temperature controller is saying.
Don't know if this helps you.
As said earlier, setting pwm=0 may not really stop your fan. This depends on the fan. If your fan keeps spinning at pwm=0, there is nothing you can do. You would need to power it off.

Maybe you find also helpful
- standby button on screen
- pure "fan mode", without temperature control. Discovery is also supported (https://www.home-assistant.io/integrations/fan.mqtt/)
- new file "config_override.h". Almost all the changes you need to do in the settings can now be done in that file

s4saken, 2023-11-27:
I am still working on this project, and i am wondering if you want another pull request when i am done?
I am going for clean configuration following most of your example. Also i am learning a bunch. I am also running code inspection from the platform regularly.

At this moment the following has been added and tested successfully and is fully compatible with fan_only/off mode:

- Added a second/third pwm channel without tacho, setup through config.h.
  >> MQTT >> stat/cmd >> adjustable via mqtt.
- Added pwm offset value for second fan, setup through config.h.
 >> MQTT >> stat/cmd >> adjustable via mqtt.
- Added relays.h/relays.cpp >> created 2 relaychannels, setup through config.h.
>> MQTT >> stat/cmd >> adjustable via mqtt.

Currently working on:
- Adding function in temperaturecontroller for switching relay on when settemperature is higher then actualtemperature, for example, a heater.
- Adding Ha Autodiscovery for other pwm channels, although i am not yet sure on how i want to set this up. Might be through mqtt fan.

Just let me know.

KlausMu, 2023-11-27:
I think the project should focussing on cooling. That's the main purpose of fans (beside reducing moisture).
So supporting a relay for heating is out of scope, I think.

But what could be of interest is having more than one pwm channel, including pwm offset for the other fans.
But they would all be controlled by the same temperatureController, right (the only difference is the pwm offset)?
Or would you also introduce several temperatureControllers? That would be rather complex, and I fear that it would not longer be possible to understand the code easily.

Regarding HA discovery, I think we would still have only one single climate or fan device (the latter setting pwm as percentage). For each fan we would have two dedicated sensors for pwm (absolute value) and rpm.

s4saken, 2023-11-27:
> So the "in" fan runs at higher pwm as the "out" fan, the result is overpressure, right? Ok, now I think I got your idea.

Yes.

s4saken, 2023-12-28:
Hi Klaus, 

Wanted you to know i don't think i can create the function in the code. 
Atm my code has been so heavily modified, i don't think i will be able to go back to the main repo. 

Just wanted to thank you again for the work and wanted to tell you what i created, its finished. 

Fan
- I created 4 pwm channels. Fan1, Fan2-offset, Fan-heater, Fan-filter.
Modified static code for curve points, min, max, to listen to cmd topics for the settings. Created discovery messages to add these to home assistant, through number component. I created a different mode for fan filter, this is added under fan component, so i can be set in speed like a fan.

Relais:
- I created 3 relais channels. Light, Heater, Power. Including modes on/off for all of them.
These are added to home assistant through switch component. I later modified the temperature controller to add heating function on/off mechanic, through min max from target.  I also made all of these values, setable through mqtt command topic, also adding them through number component with mqtt. 

Finally i choose to add heater mode to hvac, so it all comes together nicely. To my suprise the esp32, did not get slow at all. Touch also did not get slow, which is amazing in my opinion. 

I don't use the retained messages for discovery messages, but i do use them for status topics, and settings topics. This makes using the controller very easy. Using the controller with stock values is possible, if setup through mqtt, the mqtt server will act as a settings manager for the controller.  I tested everything, and it is working like charm, there where some challenges for me, but inspecting your code, and thinking out of the box, worked very well. 

Kind Regard, 
4Saken
[image]

KlausMu, 2023-12-28:
Looks great, thanks for the update!
Which hardware are you using for heating the printer enclosure?

s4saken, 2023-12-30:
> Looks great, thanks for the update! Which hardware are you using for heating the printer enclosure?

I first wanted to use a ceramic unit, ptc. But because they are a capacitive load, they cause some interference on the grid. I opted for this:

[ali](https://nl.aliexpress.com/item/1005004389653163.html?spm=a2g0o.productlist.main.1.773d3a20bwywna&algo_pvid=8b381bdd-1cd2-4224-8724-1bbd426924dc&algo_exp_id=8b381bdd-1cd2-4224-8724-1bbd426924dc-0&pdp_npi=4%40dis%21EUR%2137.16%2137.16%21%21%21285.20%21%21%40211b612817039428809203837e74a0%2112000029009630774%21sea%21NL%211789781824%21&curPageLogUid=AhndjrpWTYfK)

400 watts. Using a ssr relais 40a, with a heatsink. Since most of the hardware is in the box to, i created a box for all the hardware, and i am pulling some air though this. This solution saves me power also, because i dont have to keep the room on temperature.