KlausMu/esp32-fan-controller #20: Device Goes Offline and Stops Climate Control

krazyfingaz, 2024-01-22:
Hi Klaus,

Thanks for all the effort you've put into this project. I followed your excellent guide and got everything up and running.

My setup has the BME, makes use of MQTT (with authentication) and has no touch screen.

Everything works great off a fresh boot. However, the system can't seem to sustain its sanity for more than about a day. MQTT topics go dark and the fan seems to hold onto it's minimum rpm and not budge from there in response to the changing ambient temperature. A reboot makes everything good again. For a while.

I realize the above might be hard to reproduce, but do you have any idea where it might be falling off the rails or while the failure seems to be wholesale? I wouldn't mind so much if it lost connectivity to the MQTT server but continued to manage the fan.

KlausMu, 2024-01-22:
Hi, sad to hear that you faced some problems. 
I never had something similar. My device runs definitely longer than one day.
But if the problem is reproducible after one day longest, could you connect a notebook via USB to the ESP32 and keep the serial monitor running? In the log hopefully you can see what is going on.

krazyfingaz, 2024-01-22:
I'll try that over this weekend. I never ran it for very long on the PC while working on it. That setup will also change one variable in the sense that the power supply for the board will be the PC, whilst the fan will be powered by an external 12V power supply (with a common ground of course). In the 'production setup' the board gets its 5V power from a buck converter that hangs off the same 12V power supply that powers the fan.

I can vouch for the power stability of the buck converter since I use it in other ESP32 based IOT devices that have months of uptime and only go offline due to factors I can account for.

I'll let you know how I fare and share any interesting logs that I get while running tethered to the PC.

KlausMu, 2024-01-23:
Ok. It is also possible to get serial output without powering the ESP32. You had to use RX/TX for this.
You have to use a FTDI USB to TTL converter for this, which is very cheap.
Let me know if this is needed. Good luck.

closed 2024-03-12 by KlausMu