KlausMu/esp32-fan-controller #21: Switch between sensor source? Temp <> Humidity

Gittegatt, 2024-03-09:
Hello Klaus,
thank you for offering that code. Is there a possibility to add function to switch between sensor source? Means PWM control based on temperature or humidity values? Maybe with a slider to change between source?

Regards,
Daniel.

KlausMu, 2024-03-10:
Hi Daniel, thanks for your suggestion and that you tried the code.

Yes, I think this is achievable and would be an improvement.

The [Home Assistant MQTT climate](https://www.home-assistant.io/integrations/climate.mqtt/) also supports the drying mode.
Currently the esp32-fan-controller software only offers the [ modes "off" and "fan_only"](https://www.home-assistant.io/integrations/climate.mqtt/#modes)

I will put it on my ToDo list.

Gittegatt, 2024-03-10:
Great Klaus,
or sensor value of both with a checkbox for sensor value treshold consideration to use both, temperatur and humidity?

Regards,
Daniel.

KlausMu, 2024-03-10:
What exactly do you mean?

- fan mode
-- based on temperatrue
- dry mode: user can choose between
-- based on humidity only
-- based on temperature and humidty

Well, yes, could be done, but the two dry modes couldn't be set in the Thermostat Card of Home Assistant. An additional switch would be needed. And of course, what is going on the behind the scenes, so wich PWM value at which temperature and humidity, has to be set in code anyway, see how it is done in "temperatureController.cpp".

So I don't know if this additional complexity with the two operation modes in "dry mode" is really necessary. If you want to have it, you can do it in code. Or do you really have the need to switch between the two dry modes at runtime?

BTW, if you urgently need it, you can directly change it now in "temperatureController.cpp", you don't have to wait for any change by me.