KlausMu/esp32-fan-controller #26: Betrieb mit 24v 3-Pin Lüfter

cicatrix87, 2024-08-19:
Servus Klaus,

meinst du es wäre möglich die Schaltung soweit umzubauen, dass man das Projekt auch mit einem 24V 3-Pin Lüfter nutzen könnte? Ich habe die Herausforderung einer dezentralen Lüftungsanlage die ziemlich dumm ist und auch gerne mal Funkaussetzer hat. Jetzt war die Überlegung ob ich sie selbst Smart bekommen kann. 24V DC bekomme ich von der originalen Platine abgegriffen, den ESP32 könnte man per StepDown Driver versorgen.

Grüße Sebastian

KlausMu, 2024-08-19:
Du musst schauen, dass die Spannungen auf dem pwm pin und tacho pin zusammenpassen. Siehe auch hier https://github.com/KlausMu/esp32-fan-controller/wiki/02-Obstacle-ESP32-electronics#5v-for-fan-pwm-and-tacho-vs-33v-for-esp32
Dort ähnliche Fragestellung mit 5V/3.3 V
PWM kannst du einfach testen, ob der Fan auf 3.3V überhaupt reagiert.
Beim tacho musst du aufpassen. Im schlimmsten Fall wird der ESP32 gegrillt, es könnte aber auch funktionieren. Level Shifter löst das Problem.

cicatrix87, 2024-08-20:
Ich muss mal gucken ob das hier wirklich kein PWM Lüfter ist, ansonsten müsste ich die Geschwindigkeit ja tastsächlich über die Eingangsspannung beim Lüfter regeln.

closed 2024-10-05 by KlausMu