KlausMu/esp32-fan-controller #27: Home Assistant - Fan Controller after some time 'not availible'

richie241, 2024-09-19:
I have the following problem:
After a few hours, the 'Fan Controller' control is no longer available in HA. The fan sensors continue to run and work.
HA + MQTT latest version, hosted on a Synology 923+

[image]

KlausMu, 2024-09-19:
Could you run an additional MQTT client (e.g. MQTT Explorer), keep it running and watch what is going on?

The topic of interest is "esp32_fan_controller/stat/STATUS"
Payload "online" is sent when the fan controller powers up.
Payload "offline" is sent when the fan controller disconnects from the mqtt broker.
In MQTT explorer, in the history you see when which payload was sent.
[image]

[image]

Maybe the connection between the fan controller and the MQTT broker was lost, and after reconnection "online" was not sent again. You could check in the history of the sensors if you had continous updates or if you see a gap there (which could mean the connection was lost at some point).

KlausMu, 2024-09-21:
I am pretty sure that the connection to the mqtt broker was lost at some point, and was recovered later. This would explain why sensors got updated again, but not the online status.

I just updated the source with a simple bugfix, to send the update status again after a reconnect.
https://github.com/KlausMu/esp32-fan-controller/commit/e20e0c3fef71c5be7a94b174af962c7f064f544c

Could you please check an try this version?

richie241, 2024-09-21:
Thanks, i had a look with MQTT Explorer:

fline
21.09.2024 16:56:38 
offline
21.09.2024 16:55:59(-38.43 seconds) 
offline
21.09.2024 16:54:47(-72.19 seconds) 
offline
21.09.2024 16:45:21(-9.44 minutes) 
offline
21.09.2024 16:15:56(-29.42 minutes) 
offline
21.09.2024 15:06:20(-69.6 minutes) 
online

I test the new Version now and give you feedback :-)

KlausMu, 2024-10-08:
Hi @richie241 , any update for us?
Do you still get "not available"? Or do the reconnects work?

closed 2024-10-13 by KlausMu

richie241, 2024-10-13:
Sorry for the late reply. We were on vacation. The change went through without any errors during that time, everything seems to be working. Thank you very much.