KlausMu/esp32-fan-controller #29: PlatformIO errors:  [.pio/build/esp32dev/src/fanPWM.cpp.o] Error 1

pflugj, 2025-01-05:
Hi.
I am getting the following errors when trying to build with PlatformIO:

Compiling .pio/build/esp32dev/lib560/Wire/Wire.cpp.o
src/fanPWM.cpp: In function 'void initPWMfan()':
src/fanPWM.cpp:17:3: error: 'ledcSetup' was not declared in this scope
   17 |   ledcSetup(PWMCHANNEL, PWMFREQ, PWMRESOLUTION);
      |   ^~~~~~~~~
src/fanPWM.cpp:19:3: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   19 |   ledcAttachPin(PWMPIN, PWMCHANNEL);
      |   ^~~~~~~~~~~~~
      |   ledcAttach
Compiling .pio/build/esp32dev/liba0c/SPI/SPI.cpp.o
*** [.pio/build/esp32dev/src/fanPWM.cpp.o] Error 1
Compiling .pio/build/esp32dev/lib055/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
src/mqtt.cpp: In function 'void callback(char*, byte*, unsigned int)':
src/mqtt.cpp:258:25: error: variable 'std::string strPayload' has initializer but incomplete type
  258 |   std::string strPayload(reinterpret_cast<const char *>(payload), length);
      |                         ^
*** [.pio/build/esp32dev/src/mqtt.cpp.o] Error 1

Any idea what I have to do here??

Thanks for helping

KlausMu, 2025-01-05:
For me it looks like something is broken with your PlatformIO installation.
I just tested it with the Github action and with a local build. Both work without any error.

Is your PlatformIO installation special in any way? Did it work before?

Are you able to compile a very basic example using the missing functions mentioned above, `ledcSetup` and `ledcAttachPin`?
Both are in `platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-ledc.c` and should be found.

pflugj, 2025-01-06:
Hi.
I am running Visual Studio & PlatformIO on Kubuntu 24.04LTS.
I have never used Visual Studio & PlatformIO before.
It installed the Visual Code extension "PlatformIO IDE  v3.3.3"

Maybe there is is some clue here:
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html
Removed APIs
    ledcSetup
    ledcAttachPin

ledcAttach used to set up the LEDC pin (merged ledcSetup and ledcAttachPin functions).

But I would not know how to fix this :-(

...any ideas?
thanks for helping

pflugj, 2025-01-06:
I applied a change to:  fanPWM.cpp
ledcSetup(PWMCHANNEL, PWMFREQ, PWMRESOLUTION);
and ledcAttachPin(PWMPIN, PWMCHANNEL);
to  
ledcAttachChannel(PWMPIN, PWMFREQ, PWMRESOLUTION, PWMCHANNEL);

Now I only get this error:
Compiling .pio/build/esp32dev/src/mqtt.cpp.o
src/mqtt.cpp: In function 'void callback(char*, byte*, unsigned int)':
src/mqtt.cpp:258:25: error: variable 'std::string strPayload' has initializer but incomplete type
  258 |   std::string strPayload(reinterpret_cast<const char *>(payload), length);
      |                         ^
... any idea how to fix this?

pflugj, 2025-01-06:
I did some searching and it looks like
#include <sstream> 
had to be added to mqtt.cpp
now there is another error:

sh: 1: Syntax error: Unterminated quoted string
*** [.pio/build/esp32dev/firmware.elf] Error 2

I can not even find that file

KlausMu, 2025-01-06:
Seems you are using Arduino core 3.0 from Espressif, which is not supported by PlatformIO. PlatformIO only supports Arduino core 2.0. https://github.com/platformio/platform-espressif32/issues/1225

How did you install everything? There are several tutorials out there. Did you follow one of these? And this ended up in having Arduino core 3.0? Did you manually install Arduino core 3.0?

pflugj, 2025-01-06:
I had VS  already installed.

I now did a complete UNINSTALL of VS and deleted all files in
HOME/.config/Code and ~/.vscode

I reinstalled and only added PlatformIO.

There are now only the following extensions:
C/C++ for Visual Studio Code
and
PlatformIO

unfortunately the problem is the same as before :-(

pflugj, 2025-01-06:
After some digging I found this:
**platform = espressif32 @ 6.9.0**
This does not throw the "ledc*" errors anymore 
but still I get the other error:
sh: 1: Syntax error: Unterminated quoted string
*** [.pio/build/esp32dev/firmware.elf] Error 2

KlausMu, 2025-01-06:
Ok, just to be sure, I created a Linux VM, installed VS, the PlatformIO extension, cloned this repo and compiled. Works.

On the first few lines of the compiler output, I see
```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
```
This is exactly what I can see in the action. So both work. I can't see any error in this repo.

Don't know what is going wrong. Which package is shown in your compiler output?
And which version do you see in your folder `.platformio/packages`

pflugj, 2025-01-06:
1st I uninstalled PlatformIO, 
then  deleted .platformio/. 
then reinstalled

This time I didn't need the: 
platform = espressif32 **@ 6.9.0** setting
It is now unchanged!
 
This is the compiler output:
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
 
Still getting this error:
Building in release mode
Linking .pio/build/esp32dev/firmware.elf
sh: 1: Syntax error: Unterminated quoted string
*** [.pio/build/esp32dev/firmware.elf] Error 2
 
I had never before installed PlatformIO via command line.

Sorry for all the trouble and thanks for your efforts

pflugj, 2025-01-06:
I believe I found the culprit :-)

Somehow the project folder name started with a  "space" char.
I added the new project folder and now it compiles without any errors :-)

What I do not understand is, the miss-leading error message!

Greetings and thanks for your efforts and help

closed 2025-01-06 by KlausMu