KlausMu/esp32-fan-controller #30: static assertion failed

jojkopeter, 2025-01-26:
Hello, thank you very much for sharing it. I'm trying to get it to work, unfortunately without success. An error is displayed with the preset #define fan_controlledByTouch: static assertion failed: "#define useOTAUpdate" is only possible with "#define useWIFI".
In addition, the Az-Delivery touch display only lights up white and there is no other display on the display. Can you please help me?

KlausMu, 2025-01-27:
Hi, if you have an assertion failure, then you obviously changed some settings in a way that is not allowed. This is what the asserts are used for: to check if the settings have some errors (e.g. if you say "use OTA", but "don't use Wifi". This is not possible, because OTA needs WiFi).

So the question is: what exactly did you change in file `config.h`?

jojkopeter, 2025-01-27:
Hello, thank you very much for your feedback. In the config.h I commented out the preset for touch only and then commented out the standard preset. I entered WiFi in config-override.h.Viele Grüße Peter Jojko

jojkopeter, 2025-01-27:
Hi, thank you again for your feedback. Here is the config.h where the problems come from. Ask for your support.

Vielen Dank
Peter Jojko

￼

> Am 27.01.2025 um 09:40 schrieb Klaus Musch ***@***.***>:
> 
> Hi

jojkopeter, 2025-01-27:
Hello, I also have problems with the Az-Touch display. I connected it according to your wiring diagram but it only lights up white. Can you please help me? I would also like to use the control on my 3D printer. Please excuse my English. It's a translator. Can you maybe speak German?
Best regards
Peter Jojko

￼

KlausMu, 2025-01-27:
Now I got what you mean. There was a bug in `config.h` if you don't use MQTT. Seems everyone is using MQTT ...
Please pull the latest version and try again.

KlausMu, 2025-01-27:
> Hello, I also have problems with the Az-Touch display. I connected it according to your wiring diagram but it only lights up white.

Did you try both settings for the old and the new AZ-touch?
The pins have to be defined differently, based on if you are having the old or the new version of AZ-touch.
Please see lines 385-406 in `config.h`

new version
```
#define LED_ON           HIGH          // override it in file "config_override.h"
#define TOUCH_CS         GPIO_NUM_14   // override it in file "config_override.h"
#define TOUCH_IRQ        GPIO_NUM_27   // override it in file "config_override.h"
```
or old version
```
#define LED_ON           LOW          // override it in file "config_override.h"
#define TOUCH_CS         GPIO_NUM_14  // override it in file "config_override.h"
#define TOUCH_IRQ        GPIO_NUM_2   // override it in file "config_override.h"
```
If it still does not work, please try some completely different example, e.g. one provided by AZ themselves, just to see if the display works at all and which settings are used in that example.

jojkopeter, 2025-01-27:
Hi, thank you for your support. The display turns white on GPIO27 and GPIO2. I have tried several displays of the same type, unfortunately without success. Do you perhaps have an idea of ​​what could be done?
Best regards
Peter

KlausMu, 2025-01-27:
I don't believe this is a problem specific to this project.
Have you tried any other example? Maybe with support from AZ delivery?
If you have a running example, you can have a look there which pins are used.
Start with an example as simple as possible and proceed from there.

jojkopeter, 2025-01-27:
Hello, I have a test firmware from Az-Delivery. The display works. The Az-Delivery firmware is a .bin file. How can I open the file to read the settings?
Can you please help me?

Best regards
Peter

jojkopeter, 2025-01-27:
> You need the source code for the firmware. Where did you download the firmware?

https://cdn.shopify.com/s/files/1/1509/1638/files/AZ-Touch_TestFirmware.zip?v=1619606879￼	
AZ-Touch_TestFirmware
ZIP-Archiv · 682 KB

KlausMu, 2025-01-27:
Do you mean one of these blogposts?
https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/internet-radio-mit-dem-esp32
https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/internetradio-mit-esp32-und-max-98357a
https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/internet-radio-mit-dem-esp32
Well, it is really hard for me to tell you what the problem is.

BTW, you checked the serial output, right? Is there anything of interest in the serial log? Or is everything "normal" and you don't see any error messages in the log? Do you see anything in the log?

When I do a google search for "white display az touch" or "az touch weißer bildschirm", I get several hits. Did you follow them?

What I would do if I were you:
- Try to create or find some very simple examples on how to use the display
- first start with the display only, no touch
- if display works, add touch functionality
- if display works with touch, compare the code with the exp32-fan-controller code and look if there are any differences
I don't believe (but don't know for sure) that AZ delivery again changed the hardware.
My guess is that the problem is something else.

Sorry, but I don't know any better advice.

jojkopeter, 2025-01-27:
Thank you. There is no problem in the protocol.
Best regards
Peter

jojkopeter, 2025-01-28:
Hi, I managed to make it work. In the config.h I commented out #define DRIVER_ILI9341. I still have one problem: with setting #define fan_controlledByTouch the display is dark. The display only works with settings with MQTT. Also the display goes off when I type outside. At the same time the switch for the fan in Homeassistant is switched on. If I switch the switch for the fan to off in Homeassistant, the display goes on. Can you please help me with this?
Best regards
Peter

KlausMu, 2025-01-28:
You commented `#define DRIVER_ILI9341` (you are not using it) and it works? Which driver are you using then, if not ILI9341? Can't believe this ... The AZ touch only works with ILI9341.

If you are using preset `fan_controlledByTouch` or `fan_controlledByMQTTandTouch`, then `#define DRIVER_ILI9341` ist automatically activated in  line 66.

What do you mean with `the display goes off when I type outside`? Outside of what are you typing?

If your screen is on when it should be off, and off when it should be on, then probably your defintion of `#define LED_ON           HIGH` is wrong.
See lines 391 and 401. Try to switch from HIGH to LOW or the other way round.

I still have the suspicion that something is messed up in your config.h ...

You can try to post your config.h here, but don't answer via mail. In that case, the attachment will be removed. Post it on github.com

jojkopeter, 2025-01-28:
Hi, there is a misunderstanding. I did not comment on the driver. It was commented, that's why it didn't work.
Here is the config.h

/*
  Before changing anything in this file, consider to copy file "config_override_example.h" to file "config_override.h" and to do your changes there.
  Doing so, you will
  - keep your credentials secret
  - most likely never have conflicts with new versions of this file
  Any define in CAPITALS can be moved to "config_override.h".
  All defines having BOTH lowercase and uppercase MUST stay in "config.h". They define the mode the "esp32 fan controller" is running in.
*/

#ifndef __CONFIG_H__
#define __CONFIG_H__

#include <driver/gpio.h>
#include <esp32-hal-gpio.h>

// --- Begin: choose operation mode -------------------------------------------------------------------------------------------------------------------------------
// ----------------------------------------------------------------------------------------------------------------------------------------------------------------
// You have two ways to choose the operation mode:
// 1. either use one of the presets
// 2. or define every low level option manually
// Recommendation is to start with one of the presets.
// In both cases, after choosing the operation mode, go further down in this file to set additional settings needed for the chosen mode (unused options should be greyed out).

#define usePresets

#if defined(usePresets)
// --- Way 1 to choose the operation mode: choose one of the presets. All further options are automatically set -----------------------------
/* These are the presets:
   Fan mode: speed of the fan will be directly set, either via mqtt, via a touch display or both
   Climate mode: speed of the fan will be controlled by temperature. If actual temperature is higher than target temperature, fan runs at high speed.
     Of course, actual temperature can never be lower than the air temperature which is transported by the fan from "outside" to "inside".
     Actual temperature: the actual temperature can be measured by an BME280 or can be provided via mqtt
     Target temperature: the target temperature will be tried to reach. The target temmperate can be provided via mqtt, via a touch display or both.
*/
// --- Begin: list of presets. Choose exactly one. ---
//#define fan_controlledByMQTT
//#define fan_controlledByTouch
#define fan_controlledByMQTTandTouch
//#define climate_controlledByBME_targetByMQTT
//#define climate_controlledByBME_targetByTouch
//#define climate_controlledByBME_targetByMQTTandTouch
//#define climate_controlledByMQTT_targetByMQTT
//#define climate_controlledByMQTT_targetByMQTTandTouch
// --- End: list of presets --------------------------

// --- based on the preset, automatically define other options --------------
// --- normally you shouldn't change the next lines -------------------------
#if defined(climate_controlledByBME_targetByMQTT) || defined(climate_controlledByBME_targetByTouch) || defined(climate_controlledByBME_targetByMQTTandTouch) || defined(climate_controlledByMQTT_targetByMQTT) || defined(climate_controlledByMQTT_targetByMQTTandTouch)
  #define useAutomaticTemperatureControl
  #if defined(climate_controlledByBME_targetByMQTT) || defined(climate_controlledByBME_targetByTouch) || defined(climate_controlledByBME_targetByMQTTandTouch)
    #define setActualTemperatureViaBME280
    #define useTemperatureSensorBME280
  #endif
  #if defined(climate_controlledByMQTT_targetByMQTT) || defined(climate_controlledByMQTT_targetByMQTTandTouch)
    #define setActualTemperatureViaMQTT
  #endif
#endif
#if defined(fan_controlledByMQTT) || defined(fan_controlledByMQTTandTouch) || defined(climate_controlledByBME_targetByMQTT) || defined(climate_controlledByBME_targetByMQTTandTouch) || defined(climate_controlledByMQTT_targetByMQTT) || defined(climate_controlledByMQTT_targetByMQTTandTouch)
  #define useWIFI
  #define useMQTT
  #define useOTAUpdate
  // #define useOTA_RTOS     // not recommended because of additional 10K of heap space needed
#endif
#if defined(fan_controlledByTouch) || defined(fan_controlledByMQTTandTouch) || defined(climate_controlledByBME_targetByTouch) || defined(climate_controlledByBME_targetByMQTTandTouch) || defined(climate_controlledByMQTT_targetByMQTTandTouch)
  #define useTFT
  #define DRIVER_ILI9341       // e.g. 2.8 inch touch panel, 320x240, used in AZ-Touch
  #define useTouch
[... 410 lines cut ...]
    (defined(fan_controlledByMQTTandTouch)                 && defined(climate_controlledByBME_targetByMQTTandTouch))  || \
    (defined(fan_controlledByMQTTandTouch)                 && defined(climate_controlledByMQTT_targetByMQTT))         || \
    (defined(fan_controlledByMQTTandTouch)                 && defined(climate_controlledByMQTT_targetByMQTTandTouch)) || \
                                                                                                                         \
    (defined(climate_controlledByBME_targetByMQTT)         && defined(climate_controlledByBME_targetByTouch))         || \
    (defined(climate_controlledByBME_targetByMQTT)         && defined(climate_controlledByBME_targetByMQTTandTouch))  || \
    (defined(climate_controlledByBME_targetByMQTT)         && defined(climate_controlledByMQTT_targetByMQTT))         || \
    (defined(climate_controlledByBME_targetByMQTT)         && defined(climate_controlledByMQTT_targetByMQTTandTouch)) || \
[... cut ...]
    (defined(climate_controlledByBME_targetByMQTTandTouch) && defined(climate_controlledByMQTT_targetByMQTTandTouch)) || \
                                                                                                                         \
    (defined(climate_controlledByMQTT_targetByMQTT)        && defined(climate_controlledByMQTT_targetByMQTTandTouch))
static_assert(false, "You cannot choose more than one preset at the same time");
#endif

#endif /*__CONFIG_H__*/

Here is also a video about the behavior of Toutch and Mqtt

https://github.com/user-attachments/assets/126eb05e-02f7-439c-b740-6560bb4d56fa

KlausMu, 2025-01-28:
The format of the file is completely broken. Cannot compare it the the original file. Please provide a download link, e.g. to OneDrive, Google Drive or something similar.

Did you try what I suggested about `#define LED_ON           HIGH`?

jojkopeter, 2025-01-28:
Hi,
Sorry. I don't really know much about it yet. I tried with High and LOW, unfortunately without success.

config.h

https://drive.google.com/file/d/1C7IJhAqZl2N6OW-2i9MvZ9ZtX-5Di8st/view?usp=sharing

KlausMu, 2025-01-28:
Ok, the only relevant thing you changed in config.h is
```
    // --- choose which display to use. Activate only one. -----------------------------------------------
    // #define DRIVER_ILI9341       // 2.8 inch touch panel, 320x240, used in AZ-Touch
    #define DRIVER_ST7735        // 1.8 inch panel,       160x128
```
That's suprising, since the AZ touch until now was using DRIVER_ILI9341, as you can see in the comment in line 106

Can you have a look at the backside of the tft if it is really a ST7735 and not a ILI9341?

You cannot simply use the DRIVER_ST7735 definition with the AZ touch. A lot of things depend on this definition. PIN definitions, screen resolution and so on. DRIVER_ST7735 will not work the the AZ touch.
What happens if you simply use this, as written in the comments?
```
    // --- choose which display to use. Activate only one. -----------------------------------------------
    #define DRIVER_ILI9341       // 2.8 inch touch panel, 320x240, used in AZ-Touch
    // #define DRIVER_ST7735        // 1.8 inch panel,       160x128

```

jojkopeter, 2025-01-28:
In line 106 I set //#define DRIVER_ILI9341 again. But the behavior is itself. The display doesn't work correctly

KlausMu, 2025-01-28:
Ok, hardware is ILI9341. So you have to use
```
    // --- choose which display to use. Activate only one. -----------------------------------------------
    #define DRIVER_ILI9341       // 2.8 inch touch panel, 320x240, used in AZ-Touch
    // #define DRIVER_ST7735        // 1.8 inch panel,       160x128

```
but you had this, which is wrong
```
    // --- choose which display to use. Activate only one. -----------------------------------------------
    // #define DRIVER_ILI9341       // 2.8 inch touch panel, 320x240, used in AZ-Touch
    #define DRIVER_ST7735        // 1.8 inch panel,       160x128
```

jojkopeter, 2025-01-28:
Thanks, I tried that too. Unfortunately it doesn't work properly. The display behaves like in my video. I'm desperate. Could it be that the wiring is wrong? But I've already tried it several times on the breadboard and it hasn't worked. I would be very grateful if you could help me.

jojkopeter, 2025-01-28:
Ok, you mean in the config.h?

jojkopeter, 2025-01-28:
So,
I commented the code with LED_ON in config.h, config_override.h and tft.cpp (//). The display lights up constantly, but has no touch function. With #define fan_controlledByTouch the display doesn't light up at all. It looks like mqtt needs to be defined. Does that help you?

KlausMu, 2025-01-28:
It does not make sense to change something both in `config.h` and `config_override.h`, because every setting from `config_override.h `will overwrite the setting in `config.h`, as the filename says ...

Ok, if you have a `config_override.h` then change the line
```
#define LED_ON           HIGH          // override here
```
to
```
#define LED_ON           LOW          // override here
```
An try to understand how `config.h` and `config_override.h` work. If you don't understand it, you don't understand the changes that you are doing in the config files and if they really work or not.

jojkopeter, 2025-01-29:
Hi, it's me again. So, the display is now lighting up properly, with the LED_ON LOW setting. Unfortunately, the touchscreen is not working. Only data is shown, but I can't control the fan with "+" or "-" because it doesn't work. The button to turn it off doesn't work either. What am I doing wrong now?
Best regards
Peter

KlausMu, 2025-01-29:
Did you realize that there are two set of settings for the AZ touch, one for the old version and one for the new version of the hardware?
```
// Only AZ-Touch: here you have to set the pin for TOUCH_IRQ. The older "ArduiTouch" and the newer "AZ-Touch" use different pins. And you have to set the LED-PIN to different values to light up the TFT.
// 1. "ArduiTouch" 2.4 inch (older version)
// https://www.az-delivery.de/en/products/az-touch-wandgehauseset-mit-touchscreen-fur-esp8266-und-esp32
#ifdef useTFT
// #define LED_ON           LOW          // override it in file "config_override.h"
#endif
#ifdef useTouch
// #define TOUCH_CS         GPIO_NUM_14  // override it in file "config_override.h"
// #define TOUCH_IRQ        GPIO_NUM_2   // override it in file "config_override.h"
#endif
// 2. "AZ-Touch" 2.8 inch, since November 2020
// https://www.az-delivery.de/en/products/az-touch-wandgehauseset-mit-2-8-zoll-touchscreen-fur-esp8266-und-esp32
// https://www.az-delivery.de/en/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/az-touch-mod
#ifdef useTFT
#define LED_ON           HIGH          // override it in file "config_override.h"
#endif
#ifdef useTouch
#define TOUCH_CS         GPIO_NUM_14   // override it in file "config_override.h"
#define TOUCH_IRQ        GPIO_NUM_27   // override it in file "config_override.h"
#endif
```
If you have to use `#define LED_ON           LOW`, did you also try `#define TOUCH_IRQ        GPIO_NUM_2`?

Please read carefully file `config.h` and do your changes in file `config_override.h`

If it is still not working, please provide a full log, from the start of the ESP32 until you did some touches on the screen.

jojkopeter, 2025-01-29:
I have read the config.h carefully several times. I have the 2.8" display, the new version, and with this version pin 27 should be set. I have now set pin 2 in the config-override.g anyway and the touchscreen doesn't work either. I noticed that the display is fluttering. The display is OK though, because I tested it with Az-Delivery Test. Can you please tell me how I can create the protocol (log)?

https://github.com/user-attachments/assets/781af875-80d2-4afa-99e2-f80f49cb64d7

jojkopeter, 2025-01-29:
I found out what the error was:
In the tft.cpp file I swapped line 291 with line 287. Now the display can run with LED_ON High with PIN27. As far as the touch is concerned, the touch areas are not correct. Touch works on the opposite side. How can I configure it so that the touch works when I press the icon?

KlausMu, 2025-01-30:
Try to change `#define TFT_ROTATION` You can use 1 or 3. See comments in config.h

With log I mean the normal serial log of the ESP32. You can get it in Visual Studio Code with the PlatformIO plugin.

jojkopeter, 2025-01-30:
I've already tried TFT_ROTATION but it doesn't work. Only the display is rotated but the toutch function is not. Are there any other settings?

KlausMu, 2025-01-30:
No setting, but you can easily adjust the code here:
https://github.com/KlausMu/esp32-fan-controller/blob/956c17739a874f169e8793a02ca05295e6212361/src/tftTouch.cpp#L23

jojkopeter, 2025-01-30:
OK, I'll try it tonight. Another question: Is there a way to stop the display from fluttering? eliminate?

https://github.com/KlausMu/esp32-fan-controller/blob/956c17739a874f169e8793a02ca05295e6212361/src/tftTouch.cpp#L23

KlausMu, 2025-01-30:
The flickering is "normal". It is the update of the screen which happens every second.
For me this was always acceptable.
Could most likely be avoided by using double buffering. You would have to add this to the code by yourself.

jojkopeter, 2025-01-30:
Hi,
I would like to thank you from the bottom of my heart for your help and patience. The display is now working properly. I would like to set up an OTA update, and when I enter the IP address of the esp32 in platforio.ini I get the following error message when flashing:
[ERROR]: No response from the ESP

The IP address is correct though. Is there anything else I need to watch out for?

KlausMu, 2025-01-30:
You need to use one of the presets with MQTT support. Only in that case WiFi and OTA are available.
Check that `#define useOTAUpdate` is enabled in config.h and that you see `OTA Initialized` in the serial output log of the ESP32 on startup.

And you need to send an MQTT message "esp32_fan_controller/cmnd/OTA" with payload "ON" to the device to set it to OTA mode. See file mqtt.cpp and the documentation
https://github.com/KlausMu/esp32-fan-controller/wiki/07-OTA---Over-the-air-updates

Watch the log if the ESP32 received this message.

jojkopeter, 2025-01-30:
here is my config.h

https://drive.google.com/file/d/1C7IJhAqZl2N6OW-2i9MvZ9ZtX-5Di8st/view?usp=sharing

jojkopeter, 2025-01-30:
Can you please tell me how I get the log file in platformio?
How do I send it? MQTT message "esp32_fan_controller/cmnd/OTA"

KlausMu, 2025-01-31:
I am really glad that you got it running.

Please allow me to say that you would greatly benefit if you went through some tutorials about PlatformIO IDE in Visual Studio Code. Your last questions are not specific about this project, but could easily answered by yourself if you did a google research. There are plenty of very good tutorials.

The serial log is the most important way to get information from a running ESP32. If you have never heard of it before, you are missing something very fundamental.
https://docs.platformio.org/en/latest/integration/ide/pioide.html
You can access the serial monitor here:

[image]

You could start reading about MQTT here:
https://www.hivemq.com/mqtt/
I am using https://mqtt-explorer.com/

jojkopeter, 2025-01-31:
Thank you for trying to help me, but unfortunately I can't get the OTA to work. Maybe you have my settings in the config file. h looked at? Is everything set up correctly for OTA?

jojkopeter, 2025-01-31:
Here is a screenshot from Platformio

[image]

KlausMu, 2025-01-31:
`config.h` is correct.
Sorry, I am afraid I have to give up.

Unfortunately you don't tell me what you see in the serial log
And you don't tell me if you successfully put the ESP32 in OTA mode via mqtt.

You can use serial upload anyway. No need to use OTA.

jojkopeter, 2025-01-31:
Hi,

[image]

jojkopeter, 2025-01-31:
I think this is my last question and request. I have installed MQTT Explore. Can you please show me how I have to enter the message esp32_fan_controller/cmnd/OTA with payload "ON"?

[image]

jojkopeter, 2025-01-31:
OK,
I figured it out and it works. Thank you for your patience.
Can I get back to you if I have any other questions?

jojkopeter, 2025-01-31:
Hello, I'm back.
I noticed that the values ​​on the display are not correct. E.g. The fan reaches 152% at 2280 rpm, but this is not possible. How can I correct this?
In addition, the fan does not stop when the target temperature is reached. Can this also be corrected?

Best regards
Peter

[image]

KlausMu, 2025-02-03:
> I noticed that the values ​​on the display are not correct. E.g. The fan reaches 152% at 2280 rpm, but this is not possible. How can I correct this?

https://github.com/KlausMu/esp32-fan-controller/blob/40cc215b182751dc8ffbd78c99727f661a617fde/src/config.h#L159C9-L159C18

> In addition, the fan does not stop when the target temperature is reached. Can this also be corrected?

If you want to adjust at how much pwm the fan should run at which temperature difference, you can do it here:
https://github.com/KlausMu/esp32-fan-controller/blob/40cc215b182751dc8ffbd78c99727f661a617fde/src/config.h#L178
or here
https://github.com/KlausMu/esp32-fan-controller/blob/40cc215b182751dc8ffbd78c99727f661a617fde/src/temperatureController.cpp#L46

If a fan stops at pwm=0, depends on the fan. Some do, some don't.

closed 2025-02-05 by KlausMu