KlausMu/esp32-fan-controller #31: tempoffset

jojkopeter, 2025-02-04:
Hello,
I would like to ask if it is possible to use a "temp offset" to correct the temperature sensor?
Best regards
Peter

KlausMu, 2025-02-05:
Sure, it is possible. You have the full source code of the software, adjust it to your needs.

The code is quite simple. Filenames and names of functions are self explaining.

You could either adjust the read values in `sensorBME280.cpp` or adjust the already mentioned `temperatureController.cpp` and add the offset there (see your other issue).

The code is very simple and easy to understand. If it is still too hard for you to understand, I would propose that you go through one of the C tutorials available in the Internet.

Please understand. I don't  have the time to support people on how to write or adjust software. There are other forums for this.

If you have a specific question, I try to help.

jojkopeter, 2025-03-01:
Hello,
I admit that I am not proficient in C programming. I've read through a lot of tutorials, unfortunately I haven't found anything suitable on how to adjust the temperature in sensorBME.cpp. Can you perhaps help me in which area of ​​the file the adjustment should take place? The temperature difference is +4.5° C.
`#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include "config.h"
#include "log.h"
#include "temperatureController.h"

#ifdef useTemperatureSensorBME280
// Standard pressure at sea level. Will be calibrated in initBME280
float calibratedPressureAtSeaLevel = 1013.25;

float lastTempSensorValues[4];

Adafruit_BME280 bme;

TwoWire I2Cone = TwoWire(0);
bool status_BME280 = 0;
#endif

void initBME280(void){
  #ifdef useTemperatureSensorBME280
  lastTempSensorValues[0] = NAN;
  lastTempSensorValues[1] = NAN;
  lastTempSensorValues[2] = NAN;
  lastTempSensorValues[3] = NAN;

  I2Cone.begin(I2C_SDA, I2C_SCL, I2C_FREQ);
  status_BME280 = bme.begin(BME280_ADDR, &I2Cone);

  if (!status_BME280) {
    Log.printf("  Could not find a valid BME280 sensor, check wiring!\r\n");
  } else {
    Log.printf("  BME280 sucessfully initialized.\r\n");
    // Calibrate BME280 with actual pressure and given height. Will be used until restart of ESP32
    calibratedPressureAtSeaLevel = (bme.seaLevelForAltitude(HEIGHTOVERSEALEVELATYOURLOCATION, bme.readPressure() / 100.0F));
    Log.printf("  BME280 was calibrated to %.1f m\r\n", HEIGHTOVERSEALEVELATYOURLOCATION);
  }
  #else
  Log.printf("    BME280 is disabled in config.h\r\n");
  #endif
}

void updateBME280(void){
  #ifdef useTemperatureSensorBME280
  if (!status_BME280){
    Log.printf("BME280 sensor not initialized, trying again ...\r\n");
    initBME280();
    if (status_BME280){
      Log.printf("success!\r\n");
    } else {
      lastTempSensorValues[0] = NAN;
      #ifndef setActualTemperatureViaMQTT
      setActualTemperatureAndPublishMQTT(lastTempSensorValues[0]);
      #endif
      lastTempSensorValues[1] = NAN;
      lastTempSensorValues[2] = NAN;
      lastTempSensorValues[3] = NAN;
      return;
    }
  }
  lastTempSensorValues[0] = bme.readTemperature();
  #ifndef setActualTemperatureViaMQTT
  setActualTemperatureAndPublishMQTT(lastTempSensorValues[0]);
  #endif
  lastTempSensorValues[1] = bme.readPressure() / 100.0F;
  lastTempSensorValues[2] = bme.readAltitude(calibratedPressureAtSeaLevel);
  lastTempSensorValues[3] = bme.readHumidity();
  #endif
}`

KlausMu, 2025-03-01:
`bme.readTemperature()` reads the temperature from the sensor. Simply add `4.5` to it

So the complete line is
```
lastTempSensorValues[0] = bme.readTemperature() + 4.5;
```